fossasia/badgemagic-firmware #38: Flash firmware over USB without entering the bootloader

kienvo, 2024-07-29:
To enter the bootloader, the battery must be disconnected and the KEY1 button must be pressed and held, while USB is being plugged in. This is so inconvenient, so a feature to flash firmware over USB without entering the bootloader would needed.

froscon2024, 2024-09-23:
Would that be possible? My idea would be to run on the closed source preinstalled firmware some reboot command and then while its booting up pressing and holding the button. Or does the closed source preinstalled firmware do not support any reboot function like for example Android phones does?

You released short time ago the FOSS version of the firmware. From the issues here i can see that the FOSS version seem to lack things like showing the battery percentage. Does your FOSS version support updating itself or at least rebooting to the bootloader without having to desolder the battery?

kienvo, 2024-09-23:
I tried many methods to enter the bootloader from the firmware but there was no luck. It seems to have something relative to the RISC-V's rings that I haven't figured out yet, it cannot jump to a higher-privilege mem region or sort of.

But I had a workaround for the FOSS version, jump to 0x00 while holding the KEY1. This will make the chip enter the bootloader, even with the battery connected. It seems like jumping to 0x00 triggers a fault, and then the chip goes to a full reboot, and if KEY1 is low it will enter the bootloader. This was already implemented in the FOSS firmware. You can try it by holding KEY1 for more than 5 seconds.

By implementing this feature we can have a flexible way of updating firmware instead of pressing and holding KEY1 every time.

froscon2024, 2024-09-25:
I flashed the FOSS firmware release v0.1 thanks to this information on all badges. When you release v0.2 probably with the battery status symbol, i would use the soldering-free method to update to v0.2.

biemster, 2026-01-08:
> [@biemster](https://github.com/biemster) Are your user experience reports from the preinstalled closed source firmware or with the badgemagic opensource firmware?
> 
> This is why i am asking: [72b3310](https://github.com/fossasia/badgemagic-firmware/commit/72b33108519de689da63386769d2208ca3b0c002)

This open source firmware. I did not test the preinstalled one at all, just updated it straight out of the box.