fossasia/badgemagic-firmware #44: Implement version recognition

mariobehling, 2024-08-22:
A mobile app should be able to recognize the version of the firmware on the hardware and provide the option to update the firmware.

kienvo, 2024-08-22:
It's already implemented in Device Information Profile: https://github.com/fossasia/badgemagic-firmware/blob/f5874d607315ff88b27414299089b8528b5bb07c/src/ble/profile/devinfo.c#L14-L15

The app should read the firmware version string by reading the value of the Firmware Revision String Characteristic (0x2A26) in the Device Information Profile (0x18A). This will be documented soon, along with the firmware versioning.

mariobehling, 2024-09-10:
Ok, good. When you document this please reference this issue.

froscon2024, 2024-09-23:
What does the preinstalled closed source firmware answer when you read out those values? 
Does it even have a version?

kienvo, 2024-09-23:
> What does the preinstalled closed source firmware answer when you read out those values? Does it even have a version?

They left it default as the example in the EVT package: 'Firmware Revision'. Not just that characteristic but the whole 0x180A Service was left default. I think they don't have any reason to put more effort into this.

froscon2024, 2024-09-25:
So its easy detectable with for example the android app if its closed source or the opensource firmware because none of the foss releases from fossasia would let it default like the closed source one did.
Yes, i know that the foss version looks different, but the software that detects the firmware should also be able to know without the need of the user to choose which one is installed.

PR #53 (2024-10-21): feat: Automatically include version numbers
Resolves #44

Changes: 
- Get the version number from git
- Apply version to BLE's Device Info, USB's Serial Number, and charging animation.

## Summary by Sourcery

Automatically embed version numbers from git into various components of the system, including BLE Device Info, USB Serial Number, and charging animation. Enhance the build process to retrieve and define version information from git tags. Update the CI workflow to support versioning by fetching tags.

New Features:
- Automatically embed version numbers from git into the BLE Device Info, U [... cut]
closed 2024-10-21 by mariobehling