fossasia/badgemagic-firmware #51: Create a installation howto

froscon2024, 2024-09-22:
The general user would have of course the most simple question at the moment: How can i install the released FOSS firmware on the badge?
This question should be answered on the main page. https://github.com/fossasia/badgemagic-firmware

At the moment its completely unanswered. I had to dig into the issues to get out of this issue report what have to be done now: https://github.com/fossasia/badgemagic-firmware/issues/38

This repo at the same time did not tell what software to use to install the foss firmware. I had to search around in other repo to find some hints: https://github.com/fossasia/badgemagic-hardware?tab=readme-ov-file#platform

So it seem to be https://github.com/ch32-rs/wchisp

At the moment its currently not listed here: https://github.com/ch32-rs/wchisp?tab=readme-ov-file#tested-on
I plan to make a PR at https://github.com/ch32-rs/wchisp?tab=readme-ov-file#tested-on when i tested it myself.

kienvo, 2024-09-23:
It's funny that I just thought the flashing guide had been already on this repo. Turns out that just a [bootloader description](https://github.com/fossasia/badgemagic-firmware/blob/master/CH582.md#bootloader)

For flashing software, yes, it is `wchisp`. After installing `wchisp` and entering the bootloader, just run `make isp`. It will do the rest.

froscon2024, 2024-09-25:
You released a .bin and a .elf file in version 0.1. Because of a missing howto, i just tried out the .bin and that seem to have worked.
The wchisp is capable of flashing both file types: https://github.com/ch32-rs/wchisp?tab=readme-ov-file#usage

PR #58 (2024-10-08): chore: Update README for Installation, Usage and Development
This will close #51

## Summary by Sourcery

Update the README to include comprehensive instructions for installation, usage, and development, enhancing the documentation for users and developers.

Documentation:
- Add detailed installation instructions for setting up the firmware, including prerequisites and flashing steps.
- Include usage guidelines for transferring bitmaps over BLE, USB HID, and USB CDC ACM, and reading firmware version from the badge.
- Provide development instructions covering tools, build process, debugging met [... cut]
closed 2024-10-08 by mariobehling