fossasia/badgemagic-firmware #59: Flashed a CH582M badge and now its bricked - no OEM-backup available

froscon2024, 2024-10-11:
I got asked to flash a WCH CH582M badge a person had. Person have it some longer time ago probably from a opensource-conference. So we expected to be a fossasia-badge. It look similar, have CH582M, so the firmware got flashed.
Now the badge is bricked. Because the closed source firmware are read-only a OEM-firmware-backup does not exist.
Fossasia Version 0.1 .bin was flashed. Reflashed 3 times but no working state.

How does the current state with the 0.1 look like and how does it differ from the fossasia badges i have:
Bootloader worked like known. I disconnected battery, pressed and held the button next to the micro-USB connector and could write to the badge with wchisp.
The situation when i get some life out of it is after flash with disconnected battery. It then show some completely broken looking animation that should probably be the fossasia bootlogo.
After showing the unreadable animation its dead. Buttons does not work. They seem to be mapped different on that board or something else is wrong. It does not matter what or how often i press the buttons.
When i disconnect and reconnect the battery, the badge do not get back to life like it does on the other ones.
Just powering over usb can bring the badge to life with the wrong output.

I left the badge with battery disconnected because of its now fully unknown state. It should not kill its battery because of the state it is in now.

I expect i have to try debugging over UART, right? Is there a prebuild firmware file with UART (DEUBG=1) enabled?

kienvo, 2024-10-12:
Can you please post a picture of the back of this board and while it loads the boot logo? I think this badge has different assigned pins.

Also, is this board 11x44 LEDs? The current state of the firmware does not support any board different than 11x44.

There is no prebuild firmware with DEBUG enabled. You can manually build one, but it will not log anything other than USB's debug log. It was not mean to log

froscon2024, 2024-10-13:
[image]
[image]

Now i realize that it is a 11x55 badge. I did not know such existed. I knew about the ones with 11x44 and the ones with 12x48. So i have count 11 rows, checked about the chipset and flashed the firmware.

There is a 240fps video running the 0.1 firmware with its boot animation. The video file is too big to upload to github. So i had to upload it somewhere else: 
page 1: https://filetransfer.io/data-package/CYbwTDw8#link
page 2: https://file.io/j7s2lxKQHc24

I probably have now to develop the things for the different resolution. I would be happy if i could just use up 11x44 of the 11x55 rows but i do not know if this would somehow work and be more easy for the development part. Probably i would have to also modify the badge configuration software to make it working with 11x55 if its not possible to just ignore the more LED's and use just 11x44 of the badge.
Do you know at what code part i should start with this task?

kienvo, 2024-10-18:
I bought an 11x55 badge when preparing for GSoC and realized it's the wrong version. Here is my 11x55 badge:

[image]
[image]

It looks like some last few columns were hacked up to use Charlieplexing with fewer pins than usual. But not sure if my badge is the same as yours.

>I probably have now to develop the things for the different resolution. I would be happy if i could 
just use up 11x44 of the 11x55 rows but i do not know if this would somehow work and be more easy for the development part. Probably i would have to also modify the badge configuration software to make it working with 11x55 if its not possible to just ignore the more LED's and use just 11x44 of the badge.
Do you know at what code part i should start with this task?

To modify the firmware for your badge, first trace the LEDs according to [Charlieplexing](https://en.wikipedia.org/wiki/Charlieplexing) then change the assigned pin here:

https://github.com/fossasia/badgemagic-firmware/blob/321677cf9b152f7c80503f99abd6c2c8b4d105cf/src/leddrv.c#L71-L99

You might want to change the button pin as well:

https://github.com/fossasia/badgemagic-firmware/blob/321677cf9b152f7c80503f99abd6c2c8b4d105cf/src/button.c#L13-L14

https://github.com/fossasia/badgemagic-firmware/blob/321677cf9b152f7c80503f99abd6c2c8b4d105cf/src/button.h#L12-L17

The battery measurement pin:

https://github.com/fossasia/badgemagic-firmware/blob/321677cf9b152f7c80503f99abd6c2c8b4d105cf/src/main.c#L338

The charging pin:

https://github.com/fossasia/badgemagic-firmware/blob/321677cf9b152f7c80503f99abd6c2c8b4d105cf/src/main.c#L121-L122

https://github.com/fossasia/badgemagic-firmware/blob/321677cf9b152f7c80503f99abd6c2c8b4d105cf/src/main.c#L359

And yes, it is un-unified. I was thinking of converting these pin configs to an HAL module where we can config the pins for each version of the badge easily.

froscon2024, 2025-03-04:
I am still at this and the badge is awaiting some firmware to work with because its a more complex task to understand the part with [Charlieplexing](https://en.wikipedia.org/wiki/Charlieplexing).
I hope i could find a solution for this this to recover the badge.