fossasia/badgemagic-firmware #62: Badge can't be turned off with the top button

robinschweitz, 2024-12-30:
# Description

When compiling the latest version of the code in the master branch the compiled firmware is not working correctly. 
It should be possible to turn off the badge by pressing the top button multiple times but it always restarts instead of shutting down.

# Why is this a problem

When you flash this version of the firmware on the badge you can't shut it down anymore you would need to wait for the battery to run empty.

# Steps to reproduce

* Git clone the repo
* Compile the code
* Flash the firmware to the device
* Try turning it off

kienvo, 2025-01-01:
Hi @robinschweitz, I was trying to reproduce but not having this problem. What is your version of the badge? Was the USB receptacle connected when this problem happened? Can you please try to flash your badge with the released firmware, and check if it having the same issue?

robinschweitz, 2025-01-01:
> Hi @robinschweitz, I was trying to reproduce but not having this problem. What is your version of the badge? Was the USB receptacle connected when this problem happened? Can you please try to flash your badge with the released firmware, and check if it having the same issue?

this https://github.com/fossasia/badgemagic-firmware/pull/61 pull request fixed the issue

closed 2025-01-01 by robinschweitz