fossasia/badgemagic-firmware #65: Clarification on flashing new firmware

Xenira, 2025-01-05:
Readme states:

> Make sure the battery is disconnected.

Do I really need to desolder the battery before flashing a new firmware? Guess I have a switch that I could solder in there but would still be a little inconvenient.

kilodpace, 2025-01-06:
They mention it's because you can only enter bootloader mode when the MCU powers up. But on the boards they sell there is an unpopulated reset button, I wonder if that resets the MCU and re-enables the bootloader

Xenira, 2025-01-06:
Yeah, managed to reset as well. Hoped the other pads to be the boot button pads, but did not manage to enter bootloader either

fuero, 2025-01-24:
I desoldered the battery and followed the firmware flashing instructions.
The firmware from the release page has issues, the one from the [`bin` branch](https://github.com/fossasia/badgemagic-firmware/tree/bin/) works fine.

Be sure to also read the notes given at the [wchisp](https://github.com/ch32-rs/wchisp) page.

On the new firmware, pressing and holding the second button with cable disconnected enters the bootloader (signified by one a single lit pixel in the middle of the display).

mtojek, 2025-02-20:
@fuero As posted [here](https://github.com/fossasia/badgemagic-firmware/issues/78) my board with original firmware has the RESET button. Should I keep it pressed while pressing KEY2 or rather desolder the battery?

I'm cool with soldering but it would be inconvenient for development purposes to constantly solder/desolder the batt.

fuero, 2025-02-24:
Mine looks exactly the same, but has no RESET button soldered in. As I mentioned previously, once you have the new firmware version flashed, pressing and holding KEY2 before connecting the cable enters the bootloader *with the battery still connected*.