fossasia/badgemagic-firmware #79: Debugging method

kienvo, 2025-02-21:
I was able to debug the firmware using a WCH-linkE connected to PB14/PB15 (SWDIO/SWDCLK).

[image]

Before using the debug port:

- Enable debug port by `wchisp config enable-debug` and `wchisp config unprotect` while in the isp mode (bootloader). Firmware flash will be erased.
- Cut the trace of PB14/PB15 to the LED, and connect it to the WCH-linkE dongle. For more details see fossasia/badgemagic-hardware#31.
- Flash the firmware to `0x00000000` using `wlink` and the WCH-linkE dongle over the debug port.

Further details need to be investigated to set up OpenOCD.

froscon2024, 2025-04-08:
Debugging over USB should be also possible: https://github.com/ch32-rs/wlink/issues/69

Either 'config unprotect' or 'config enable-debug' have to be used. One of the commands could brick the badge, but its also described there how to recover and you have a presoldered badge where you could use in addition the WCH-linkE dongle.

kienvo, 2025-04-08:
If I understand correctly, wlink cannot be used with the chip's ISP mode (over USB, what is so-called the bootloader), it's meant to be used with WCH-Link* dongles to flash/debug over the debug port.