fossasia/badgemagic-firmware #80: Badge softbricking itself when switching modes

morllz, 2025-02-23:
Title explains it. Had to de- and resolder the batterie to get it back working.
Couldn't really figure out whats the cause of this and it doesn't happen everytime. Running the latest firmware.

kienvo, 2025-03-05:
It sounds like the badge enters the power-off mode while the battery is disconnected. Since no voltage is present at the KEY1 button while the battery is disconnected, it won't wake up.

froscon2024, 2025-03-13:
I can confirm this issue. I have additional information about it:
I installed the 3 weeks old binary on two micro-usb badges. After just few minutes playing around with them one was bricked. It was not able to turn it on again. It seemed fully dead. The battery was not empty. Charging or anything else over usb did not worked.
I then played around with the second badge and that also bricked after few minutes.

I then had two looking dead badges.
I put them away for about 4 days and connected them again to a charger. Suddenly both came back to life again. The battery symbol was showing that both batteries was completely empty. The battery symbol had an exclamation mark inside.

I charged them for a moment and then turned them off.

> Since no voltage is present at the KEY1 button while the battery is disconnected, it won't wake up.

I think there was voltage present at KEY1. I had not the chance to measure anything. It seem to be just really hard software bricked. And because the RISC-V chip seemed to not be in sleep mode, the battery was quickly empty.

Do you need some additional test reports? I would like to avoid soldering again on the PCB and also avoid discharging the battery (leaving it software bricked on the table) that low so that it acts like a disconnected battery. Both things are for sure not healthy for the device.

kienvo, 2025-03-18:
Happened to me also. It can't turn on again after turning off while in Bluetooth mode and charging. But works fine without charging.

froscon2024, 2025-03-19:
Because i left the firmware on the badge and was now fully sure how i got into the state, i accidental tripped it once again.
But now i know how to not trigger this issue and thus how to reproduce it:
The starting point have to be a badge that is turned off.
When you attach a usb cable to the turned off badge, then a second later you know if you trip the state where it could brick itself or not:
When you attach the usb cable to a turned off badge, the badge should go into charging mode. But when it instead just start up the way when you press the button, then you are on the bricked trail. To jump off this bricked-trail, you have to disconnect the usb cable from the badge before turning it off. Because now if you leave the usb cable attached, and press the button once, it goes to bluetooth mode. Now is the very last chance to get off this bricked-trail and you have to disconnect the usb cable now before turning it off with the next button press. If you press the button once again, then you have the hard-software-bricked state where its discharging the battery to a probably really unhealthy voltage.
I tried every few hours to bring it back to life by attaching a usb cable. It took about 48 hours on my charging state (not full) to deplate the battery that low, that i could connect the usb cable and it start again.

It could maybe help when you modify the code so that it always starts into charging mode like i think it should when a usb cable is attached while turned off. But it would be probably even better when you could debug the code so that such a hard-software-bricked state is impossible. Even now when it wrongly start into the normal mode instead of the charging mode it should never be able to get into such a bricked state.